mobizt/Firebase-ESP8266 #323: Complie issue

abh-shk, 2022-04-14:
# I am using ESP8266

as per instruction to avoid fingerprint issue i downloaded "Firebase-ESP8266-master" for which i am constantly getting error while compiling.

## I am using 
1. ArduinoJson-5.x
2. esp82666 boards installed 2.74

 My Code below:
#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>
                          
#define FIREBASE_HOST "load-abh-homeaut-default-rtdb.firebaseio.com"                     //Your Firebase Project URL goes here without "http:" , "\" and "/"
#define FIREBASE_AUTH "NJ***********************************mKb**" //Your Firebase Database Secret goes here

#define WIFI_SSID "A*******I"                                               //your WiFi SSID for which yout NodeMCU connects
#define WIFI_PASSWORD "Password123"                                      //Password of your wifi network 
 

// Declare the Firebase Data object in the global scope
FirebaseData firebaseData;

// Declare global variable to store value
int val=0;

void setup() {

  Serial.begin(115200);                                                      // Select the same baud rate if you want to see the datas on Serial Monitor

  Serial.println("Serial communication started\n\n");  
           
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);                                     //try to connect with wifi
  Serial.print("Connecting to ");
  Serial.print(WIFI_SSID);

  
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }

  
  Serial.println();
  Serial.print("Connected to ");
  Serial.println(WIFI_SSID);
  Serial.print("IP Address is : ");
  Serial.println(WiFi.localIP());                                            //print local IP address
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);   // connect to firebase

  Firebase.reconnectWiFi(true);
  delay(1000);
}

void loop() {

// Firebase Error Handling And Reading Data From Specified Path ************************************************

if (Firebase.getInt(firebaseData, "/data")) {                           // On successful Read operation, function returns 1  

    if (firebaseData.dataType() == "int") {                            // print read data if it is integer

      val = firebaseData.intData();
      Serial.println(val);
      Serial.println("\n Change value at firebase console to see changes here."); 
      delay(10000);
      
    }

  } else {
    Serial.println(firebaseData.errorReason());
[... 72 lines cut ...]
                             ^
C:\Users\Dell\Documents\Arduino\libraries\Firebase-ESP8266-master\src\json\FirebaseJson.h: At global scope:
C:\Users\Dell\Documents\Arduino\libraries\Firebase-ESP8266-master\src\json\FirebaseJson.h:1832:19: error: 'Client' has not been declared
     bool readFrom(Client &client) { return mReadClient(&client); }
                   ^
C:\Users\Dell\Documents\Arduino\libraries\Firebase-ESP8266-master\src\json\FirebaseJson.h:1834:19: error: 'Client' has not been declared
     bool readFrom(Client *client) { return mReadClient(client); }
                   ^
C:\Users\Dell\Documents\Arduino\libraries\Firebase-ESP8266-master\src\json\FirebaseJson.h:2285:19: error: 'Client' has not been declared
     bool readFrom(Client &client) { return mReadClient(&client); }
                   ^
C:\Users\Dell\Documents\Arduino\libraries\Firebase-ESP8266-master\src\json\FirebaseJson.cpp:532:36: error: 'bool FirebaseJsonBase::mReadClient' is not a static member of 'class FirebaseJsonBase'
 bool FirebaseJsonBase::mReadClient(Client *client)
                                    ^
C:\Users\Dell\Documents\Arduino\libraries\Firebase-ESP8266-master\src\json\FirebaseJson.cpp:532:36: error: 'Client' was not declared in this scope
C:\Users\Dell\Documents\Arduino\libraries\Firebase-ESP8266-master\src\json\FirebaseJson.cpp:532:44: error: 'client' was not declared in this scope
 bool FirebaseJsonBase::mReadClient(Client *client)
                                            ^
C:\Users\Dell\Documents\Arduino\libraries\Firebase-ESP8266-master\src\json\FirebaseJson.cpp:533:1: error: expected ',' or ';' before '{' token
 {
 ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

# WHAT I AM SUPPOSE TO DO TO FIX THIS ISSUE?

mobizt, 2022-04-14:
You need to update ESP8266 Arduino Core SDK. The latest version is recommended.
The version less than 2.7.4 is not supported.

closed 2022-04-14 by mobizt

abh-shk, 2022-04-15:
Thank you it really worked,

May i iknow if i can use latest ArduinoJason version also?

currently using v5.x

Lastest available v6.x

Kind Regards,
Abdullah Shaikh

mobizt, 2022-04-15:
Nothing is related to that library.

The Firebase library has better built-in FirebaseJson.