mobizt/Firebase-ESP8266 #328: Help Wanted Print Value from Firebase

Kesavan-Hex, 2022-05-08:
void automatic()
{
  ///  current Soil Level from A0
 int cursoillevel;
 cursoillevel = ( 100.00 - ( (analogRead(moistureSensor)/1023.00) * 100.00 ) );

 int lowsoillevel;
 lowsoillevel = (Firebase.getInt(fbdo, "/Smart_Farm1/Auto/Soil"));
 
 Serial.print("Current Soil Moi Level : ");
 Serial.println(cursoillevel);

 Serial.print("Current Pump Trigger Level : ");
 Serial.println(lowsoillevel);
 }

lowsoillevel always shows 1.00 in serial monitor but its different in firebase how to get correct value from firebase 
thanks in advance

mobizt, 2022-05-08:
Why don't check the usage from [example](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Basic/Basic.ino)?

closed 2022-05-08 by mobizt

mobizt, 2022-05-08:
The functions description is also available [here](https://github.com/mobizt/Firebase-ESP8266/tree/master/src).

Kesavan-Hex, 2022-05-08:
now its print correct value from the firebase by modifying code like this

void automatic()
{
  ///  current Soil Level from A0
 int cursoillevel;
 cursoillevel = ( 100.00 - ( (analogRead(moistureSensor)/1023.00) * 100.00 ) );
 ////

 
 Serial.printf("Get string... %s\n", Firebase.getString(fbdo, F("/Smart_Farm1/Auto/Soil")) ? fbdo.to<const char *>() : fbdo.errorReason().c_str());
 

 Serial.print("Current Soil Moi Level : ");
 Serial.println(cursoillevel);

 //Serial.print("Current Pump Trigger Level : ");
 //Serial.println(lowsoillevel);

 }

any example how to assign the value to an variable  (lowsoillevel)

what i have tried is 

lowsoillevel =  Serial.printf("Get string... %s\n", Firebase.getString(fbdo, F("/Smart_Farm1/Auto/Soil")) ? fbdo.to<const char *>() : fbdo.errorReason().c_str());

but this not seems to be working

mobizt, 2022-05-08:
You should have the basic Arduino and C/C++ programming.

Did you know what is the function Serial.printf do?

Please ask the basic programming question in Arduino community forum.