mobizt/Firebase-ESP8266 #339: help me, error compiling/uploading

DipaSeukesa, 2022-07-26:
error report
error: #error Architecture or board not supported.
 #error Architecture or board not supported.
  ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

firebase_ESP8266_client v 4.0.3
note esp8266 SDK v 2.7.4
arduinoJson 5x

mobizt, 2022-07-26:
This is not library issue or bugs.

If you are new to esp8266 programming, please follow the weblog or YouTube tutorials about basic esp8266 programming for how to program and compile your code.

mobizt, 2022-07-26:
The error is about you are not yet install the board packages (tool chains) or no esp,8266 based board was selected for compilation in Arduino IDE.

mobizt, 2022-07-26:
If you know how to program it correctly, the error is about you are using other libraries that don't support esp8266 which is not this library.

closed 2022-07-26 by mobizt

mobizt, 2022-07-26:
The possible cause is because SD library `SD.h` which this Firebase library included to support SD filesystem.

Normally SD.h library is in the esp8266 core library [here](https://github.com/esp8266/Arduino/blob/master/libraries/SD/src/SD.h).

When you install other SD libraries that  have the same header `SD.h` and it is not compatible with esp8266, that library may raise this error in its code. I found this error statement is from other Adafruit fork SD library.

The esp8266 core SD library is actually Adafruit fork version that adapted to use in esp8266 core.

Then you should delete all SD libraries found in your Arduino libraries folder.

DipaSeukesa, 2022-07-26:
> The possible cause is because SD library `SD.h` which this Firebase library included to support SD filesystem.
>
> Normally SD.h library is in the esp8266 core library [...]

> The possible cause is because SD library `SD.h` which this Firebase library included to support SD filesystem.
>
> Normally SD.h library is in the esp8266 core library [...]

DipaSeukesa, 2022-07-26:
Saya minta maaf untuk ini, saya hanya pengguna baru github, saya tidak tau untuk mengirim pesan pribadi ke kamu.

tapi saya bukan pemula, saya telah menyelesaikan beberapa project berbasi papan esp8266.

itu dalaha masalahnya,  dan kesalahan saya tidak melampirkan error report keseluruhan pada terminal.
terima kasi banyak .  

saya hanya ingin menggunakan librari yang kamu buat yang saya dengar bisa memperbaiki masalah certificate fingerprint firebase yang selalu berubah. apakah fitur itu masih bekerja?

> The possible cause is because SD library `SD.h` which this Firebase library included to support SD filesystem.
>
> Normally SD.h library is in the esp8266 core library [...]

DipaSeukesa, 2022-07-26:
I'm sorry for this, I'm just a new github user, I don't know how to send a private message to you.

but i'm not a newbie, i have done some esp8266 board based projects.

that was the problem, and my error was not attaching the whole error report to the terminal.
thank you very much .

I just want to use the library you created which I heard can fix the firebase certificate fingerprint issue that keeps changing. does that feature still work?

> If you know how to program it correctly, the error is about you are using other libraries that don't support esp8266 which is not this library.

DipaSeukesa, 2022-07-26:
I'm sorry for this, I'm just a new github user, I don't know how to send a private message to you.

but i'm not a newbie, i have done some esp8266 board based projects.

that was the problem, and my error was not attaching the whole error report to the terminal.
thank you very much .

I just want to use the library you created which I heard can fix the firebase certificate fingerprint issue that keeps changing. does that feature still work?

> The possible cause is because SD library `SD.h` which this Firebase library included to support SD filesystem.
>
> Normally SD.h library is in the esp8266 core library [...]

mobizt, 2022-07-26:
Don't quote my reply without doing anything. I already told you that you should delete SD library.