mobizt/Firebase-ESP8266 #340: Sending PushJSON syntax to log data always failed while streaming.

m4ri01, 2022-07-29:
Hi, I want to ask why my PushJSON syntax always fails. So, I created an IoT system to water a plant. So the trigger for watering the plant comes from the firebase value inside /message/command. To use this scenario, I stream the value from that path. Inside this system, I also want to log the time data that I got from the NTP server to firebase for every command to turn off the watering system. I want to log the data inside path/message/dataSiram with a JSON form. but every time pushJson syntax was executed, I got a "send request failed" error message. I already changed the size of BSSLBufferSize to 2048, but the result is the same. I still got an error. Anyone know why this happened? Thankyou.

PS:
this is my code:

 
```
#include <NTPClient.h>
#include <WiFiUdp.h>
#include <ArduinoJson.h>
#include <FirebaseESP8266.h>
#include "addons/TokenHelper.h"
#include "addons/RTDBHelper.h"

#ifndef STASSID
//setting wifi dan password
#define STASSID "xxxxxx"
#define STAPSK  "xxxxxxxxxxxxxx"
#endif

//alamat setting firebase dan API key
#define DATABASE_URL "xxxxxxxxxxxxxxxx"
#define API_KEY "xxxxxxxxxxxxxxxxxxxx"
FirebaseData fbdo;
FirebaseData stream;
FirebaseAuth auth;
FirebaseConfig config;
const long utcOffsetInSeconds = 25200;

WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP, "pool.ntp.org", utcOffsetInSeconds);
bool signupOK = false;
unsigned long sendDataPrevMillis = 0;
const char* ssid     = STASSID;
const char* password = STAPSK;
int i = 0;
int lastSend = 0;
int pompa =0;
String timeStr;
int message =0;
bool dataChange = false;
int lastMessage = 0;

void streamCallback(StreamData data){
  // membaca data message dari firebase
  message = data.to<int>();
  printResult(data);
  dataChange= true;
}
void streamTimeoutCallback(bool timeout)
{
if (timeout)
    Serial.println("stream timed out, resuming...\n");

  if (!stream.httpConnected())
    Serial.printf("error code: %d, reason: %s\n\n", stream.httpCode(), stream.errorReason().c_str());
}

void setup() {
  // inisialisasi komunikasi dengan arduino
  Serial.begin(9600);
  // konek wifi
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }  
  timeClient.begin();
  // Login firebase
  config.api_key = API_KEY;
  config.database_url = DATABASE_URL;
  Firebase.signUp(&config, &auth, "", "");
  
  config.token_status_callback = tokenStatusCallback;
  // konek firebase dengan esp 
  Firebase.begin(&config, &auth);
  Firebase.reconnectWiFi(true);
  #if defined(ESP8266)
    stream.setBSSLBufferSize(2048 /* Rx in bytes, 512 - 16384 */, 1024 /* Tx in bytes, 512 - 16384 */);
  #endif
    if (!Firebase.beginStream(stream, "/message/command"))
    Serial.printf("sream begin error, %s\n\n", stream.errorReason().c_str());

  Firebase.setStreamCallback(stream, streamCallback, streamTimeoutCallback);

}

void loop() {
  // ngecek jam RTC
  timeClient.update();
  int getHour = int(timeClient.getHours());
  int getMinute = int (timeClient.getMinutes());
  time_t epochTime = timeClient.getEpochTime();  
  String getFormattedTime =  String(timeClient.getFormattedTime());
  struct tm *ptm = gmtime ((time_t *)&epochTime);
  int monthDay = ptm->tm_mday;
  int currentMonth = ptm->tm_mon+1;
  int currentYear = ptm->tm_year+1900;
  String currentDate = String(currentYear) + "-" + String(currentMonth) + "-" + String(monthDay);
  // ngecek jam dan message, bila bernilai benar maka data akan dirimkan ke arduino
  if((getHour == 6 &&  getMinute == 0) || (getHour ==17 &&  getMinute == 0) || message == 1 ){
    if (lastSend == 0){
      StaticJsonDocument<200> Doc;
      Doc["message"] = 1;
      serializeJson(Doc,Serial);
      lastSend = 1;
    }
  }
  else{
    if (lastSend == 1){
    // kirim data dari firebase ke arduino
    lastSend = 0;
    StaticJsonDocument<200> Doc;
    Doc["message"] = 0;
    serializeJson(Doc,Serial); 
     if (Firebase.ready()){
        // kirim data status pompa dan jam ke firebase
        FirebaseJson parent;
        parent.add("pompa",1);
        parent.add("time",getFormattedTime);
        parent.add("tanggal",currentDate);
        Serial.printf("Set json... %s\n\n", Firebase.pushJSON(fbdo, "/message/dataSiram", parent) ? "ok" : fbdo.errorReason().c_str());
//        Serial.println("berhasil");
      }     
    }
  }
  
}
```

This is the error that i got from serial:
https://drive.google.com/file/d/18dLH6qs40RfMD-3uBes_yUqwz-dyDQM7/view?usp=sharing

mobizt, 2022-07-29:
You should know the stack and heap limits on ESP8266.

The SSL client inside the `FirebaseData` object required > 10k free memory to perform SSL/TLS handshake and it will be failed to connect to server when your device free memory is too low.

And you should avoid to do something that consumed memory and time in any callback function.

mobizt, 2022-07-29:
Follow the example and read the comment in the stream example.

closed 2022-07-29 by mobizt

mobizt, 2022-07-29:
You should enable debug options in Arduino IDE.