mobizt/Firebase-ESP8266 #342: Streaming stop after few hours

pvpatel9726, 2022-09-16:
I have try to getting value from firebase node, and firebase stream works fine for few hours after that it didn't work. there  is no error message appear on serial monitor.

#if defined(ESP32)
#include <WiFi.h>
#include <FirebaseESP32.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>
#endif

#include <addons/TokenHelper.h>
#include <addons/RTDBHelper.h>

#define WIFI_SSID "ssid"
#define WIFI_PASSWORD "password"

// For the following credentials, see examples/Authentications/SignInAsUser/EmailPassword/EmailPassword.ino

/* 2. Define the API Key */
#define API_KEY "AIzaSyDDOhEKK0lFOSpn6SY9RjGUlFHtfddfNoU"

/* 3. Define the RTDB URL */
#define DATABASE_URL "databaseUrl.firebaseio.com" //<databaseName>.firebaseio.com or <databaseName>.<region>.firebasedatabase.app

/* 4. Define the user Email and password that alreadey registerd or added in your project */
#define USER_EMAIL "emailID"
#define USER_PASSWORD "1234567890"

// Define Firebase Data objects
FirebaseData stream;

// for authentication and configuration
FirebaseAuth auth;
FirebaseConfig config;

void setup()
{
  Serial.begin(115200);
  Serial.println();
  Serial.println();
  pinMode(2,OUTPUT);
  digitalWrite(2,1);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

  
  config.database_url = DATABASE_URL;
      
  // User login details
  auth.user.email = USER_EMAIL;
  auth.user.password = USER_PASSWORD;
  
  config.api_key = API_KEY;

  config.token_status_callback = tokenStatusCallback; // see addons/TokenHelper.h

  Firebase.begin(&config, &auth);

  Firebase.reconnectWiFi(true);

  if (!Firebase.beginStream(stream, "/test/stream/data"))
    Serial.printf("sream begin error, %s\n\n", stream.errorReason().c_str());

}

void loop()
{
  if (!Firebase.ready()){
        Serial.println("firebase not ready !");
    return;
  }
  if (!Firebase.readStream(stream))
   Serial.printf("sream read error, %s\n\n", stream.errorReason().c_str());

  if (stream.streamTimeout())
  {
    Serial.println("stream timed out, resuming...\n");

    if (!stream.httpConnected())
      Serial.printf("error code: %d, reason: %s\n\n", stream.httpCode(), stream.errorReason().c_str());
  }

  if (stream.streamAvailable())
  {
    Serial.printf("sream path, %s\nevent path, %s\ndata type, %s\nevent type, %s\n\n",
                  stream.streamPath().c_str(),
                  stream.dataPath().c_str(),
                  stream.dataType().c_str(),
                  stream.eventType().c_str());
    printResult(stream); // see addons/RTDBHelper.h
    String data = stream.to<String>();
    if(data== "on") digitalWrite(2,0);
    if(data== "off") digitalWrite(2,1);
    Serial.println();
  }    
}

mobizt, 2022-09-16:
There is no bug like this in this library which has been test 24/7 for years.

When id token or access token was expired (in 59 minutes), the token will be refreshing, and you will see this debug message

```
Token info: type = id token, status = on refreshing
Token info: type = id token, status = ready
```

If the network connection is the issue, you will see the error about the connection.

closed 2022-09-16 by mobizt

pvpatel9726, 2022-09-16:
I've check the wifi connection and internet also. I have also try with different wifi router and mobile hospot, but this issue is still coming. No error message or debug message  appears in serial monitor but void loop is still running.

mobizt, 2022-09-16:
From your code, at least you should see **"firebase not ready !"** .

There is no issue for such simple code or all examples code.

You should set the debug option in Arduino IDE to see what's happen in WiFi and SSL client operation and debug it yourself.

mobizt, 2022-09-16:
If you see the stream event type **"auth_revoked"** from the last stream data, this means the id token or access token was expired and the token refreshing process is not complete before token expiry time. 

The token refreshing process will start 1 minute before your token expiry time. This may cause the problem in some situation depending how Firebase.ready was polling run.

To prevent this, you can set the time to start token refreshing process sooner by setting this in setup() like this.

https://github.com/mobizt/Firebase-ESP8266/blob/f20f0b68ee62922a709f74716598965b7c59a3f3/examples/Authentications/SignInWithIDToken/SignInWithIDToken.ino#L109-L110

mobizt, 2022-09-16:
I will update the library to refresh token sooner before expiring to prevent this `auth_revoked` issue.

pvpatel9726, 2022-09-22:
As you say I have set preRefreshSecond to 15 minutes, but still issue is not resolve. Serial monitor show the "event type, auth_revoked"  message. This message is triggered after 59 minutes of token on Request. After that stream is not working.

this is from serial monitor.

08:33:45.340 -> 
08:33:45.340 -> Connected with IP: 192.168.237.29
08:33:45.340 -> 
08:33:45.340 -> Firebase Client v4.0.5
08:33:45.340 -> 
08:33:45.340 -> Token info: type = id token, status = on request
08:33:45.340 -> [hostByName] request IP for: www.googleapis.com
08:33:45.434 -> [hostByName] Host: www.googleapis.com IP: 142.250.77.42
08:33:47.121 -> Token info: type = id token, status = ready
08:33:47.121 -> [hostByName] request IP for: rtdbDatabase.firebaseio.com
08:33:47.168 -> [hostByName] Host: rtdbDatabase.firebaseio.com IP: 34.120.160.131
08:33:49.371 -> sream path, /test/stream/data
08:33:49.371 -> event path, /
08:33:49.371 -> data type, string
08:33:49.371 -> event type, put
08:33:49.371 -> 
08:33:49.371 -> on
08:33:49.371 -> 
08:34:07.316 -> sream path, /test/stream/data
08:34:07.316 -> event path, /
08:34:07.316 -> data type, string
08:34:07.316 -> event type, put
08:34:07.316 -> 
08:34:07.316 -> off
08:34:07.316 -> 
08:40:44.361 -> wifi evt: 7
08:47:54.356 -> wifi evt: 7
08:47:54.356 -> wifi evt: 7
08:47:54.403 -> wifi evt: 7
08:47:54.403 -> wifi evt: 7
08:53:26.743 -> wifi evt: 7
08:53:26.743 -> wifi evt: 7
08:53:31.807 -> wifi evt: 7
08:53:31.807 -> wifi evt: 7
08:53:31.854 -> wifi evt: 7
09:18:40.483 -> Token info: type = id token, status = on refreshing
09:18:40.483 -> [hostByName] request IP for: securetoken.googleapis.com
09:18:40.531 -> [hostByName] Host: securetoken.googleapis.com IP: 142.250.199.170
09:18:42.278 -> Token info: type = id token, status = ready
09:19:29.978 -> sream path, /test/stream/data
09:19:29.978 -> event path, /
09:19:29.978 -> data type, string
09:19:29.978 -> event type, put
09:19:29.978 -> 
09:19:29.978 -> on
09:19:29.978 -> 
09:19:36.519 -> sream path, /test/stream/data
09:19:36.519 -> event path, /
09:19:36.519 -> data type, string
09:19:36.519 -> event type, put
09:19:36.519 -> 
09:19:36.519 -> off
09:19:36.519 -> 
09:33:39.297 -> sream path, /test/stream/data
09:33:39.297 -> event path, /
09:33:39.297 -> data type, 
09:33:39.297 -> event type, auth_revoked
09:33:39.342 -> 
09:33:39.342 -> 
09:33:39.342 ->

mobizt, 2022-09-22:
The library was update to [v4.1.0](https://github.com/mobizt/Firebase-ESP8266/releases/tag/v4.1.0), 3 days ago; did you try it?