mobizt/Firebase-ESP8266 #353: BUG id token (GITKit token), status = error

khanfar, 2023-02-02:
after upload sketch , in monitor i see these error please help : 
Token error: code: 400, message: bad request
Token info: type = id token (GITKit token), status = on request
Token info: type = id token (GITKit token), status = error
Token error: code: 400, message: CONFIGURATION_NOT_FOUND
Token info: type = id token (GITKit token), status = error
Token error: code: 400, message: bad request
Token info: type = id token (GITKit token), status = on request
Token info: type = id token (GITKit token), status = error
Token error: code: 400, message: CONFIGURATION_NOT_FOUND

mobizt, 2023-02-02:
Remove all Firebase libraries found in Arduino libraries folder and install the latest version from Library Manager.

closed 2023-02-02 by mobizt

khanfar, 2023-02-02:
What's library name to install from manager ?

mobizt, 2023-02-02:
[Firebase ESP8266 Client](https://www.arduinolibraries.info/libraries/firebase-esp8266-client)