mobizt/Firebase-ESP8266 #354: Can i use RTDB stream Callback when using ESPNOW?

ProyectoOVO, 2023-02-13:
i have tried the stream callback with the example and it works, but when i use it in my project (that has the ESPnow callback) doesnt work, it just dont trigger the function (but it can still read/write from the database, just i can make it work with the callback). What can i do to cover both functions?  

PD: It doesnt give any errors, just the function isnt triggered. Im not posting code, because is huge.

mobizt, 2023-02-14:
I don't know. 
You should debug it yourself.
You can use ping to check internet status.

closed 2023-02-14 by mobizt

ProyectoOVO, 2023-02-14:
i made the question to ask if there can be some issues to have both functions at the same time, both are callback and both are in background working, so maybe they cannot work at the same time, that should explain why it works when the stream callback is alone.

mobizt, 2023-02-14:
In ESP8266, the stream will be processed repeatedly in schedule functions then it should not have conflicts with others.

I can be memory issue.

You can use `Firebase.runStream();` in loop to execute all stream tasks manually. The result will send to the Stream callback functions. 
https://github.com/mobizt/Firebase-ESP8266/blob/ca5a18a7ce461c59e3fe23f52ca8a61018ea630c/examples/DataChangesListener/Callback/Callback.ino#L196

ProyectoOVO, 2023-02-15:
ok i solved it! it was a payload size issue, thanks you so much. I have other question, sorry for being annoying, sometimes the token generation says something like "Token error: code: -135, message: NTP server time synching failed", how can i retry until i get token status = ready by code? i would like to avoid the system to continue working until the token part is ready, otherwise maybe restart the ESP to try again

mobizt, 2023-02-16:
When server or root certificate was assigned (in ESP8266 only) or the auth method is access or custom token, library required valid time for certificate validation and jwt token signing.

If time was not set or synch when `Firebase.begin` was called, library will get the time from NTP server.   This process may fail and reports the error for let you know what's going on. The library will try to send new request to NTP sever every 10 seconds when no response from NTP server. 

The `Firebase.ready()` reurns false during this. You can set your device time using `configTime` function and wait until your device time was set before calling `Firebase.begin` then the internal time synching process will be ignored.