mobizt/Firebase-ESP8266 #356: Crashing while writing tot firebase realtime database

JoergTiedemann, 2023-03-04:
Hi
I've an application on an ESP8266 which incements a integer every 15 seconds and than writ it to a firebase realtime database 
This works, but sometime after a few seconds or minutes sometimes after a few day the esp8266 chrases and reboots and than ist starts again
the last print on serial monitor i see is shown below:

11:40:08.998 > Token info: type = id token (GITKit token), status = on refreshing
11:40:10.553 > 
11:40:10.554 > --------------- CUT HERE FOR EXCEPTION DECODER ---------------
11:40:10.559 >
11:40:10.559 > Exception (29):
11:40:10.559 > epc1=0x4023cd30 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
11:40:10.750 >
11:40:10.750 > StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
11:40:10.750 >   epc1=0x4023cd30 in HttpHelper::readLine(Client*, char*, int) at .pio\libdeps\d1_mini\Firebase ESP8266 Client\src/FB_Utils.h:1097
11:40:10.750 >      (inlined by) HttpHelper::readChunkedData(MB_FS*, Client*, char*, MB_String*, fb_esp_tcp_response_handler_t&) at .pio\libdeps\d1_mini\Firebase ESP8266 Client\src/FB_Utils.h:1228

It seems that is has something to do with writing while token is on status on refrehing ? But how can i prevent this ?
Writing will be done in the loop function as shown below:
 if (Firebase.ready() && (millis() - sendDataPrevMillis > 15000 || sendDataPrevMillis == 0)){
    sendDataPrevMillis = millis();
    // Write an Int number on the database path test/int
    if (Firebase.RTDB.setInt(&fbdo, "test/int", count)){
      Serial.println("PASSED");
      Serial.println("PATH: " + fbdo.dataPath());
      Serial.println("TYPE: " + fbdo.dataType());
    }
    else {
      Serial.println("FAILED");
      Serial.println("REASON: " + fbdo.errorReason());
    }
    count++;
  }

thanks for your help

mobizt, 2023-03-04:
That's because of memory leaks or low free heap.

You should debug your code first and this is not a library issue.

mobizt, 2023-03-04:
You should read this information
https://github.com/mobizt/Firebase-ESP8266#memory-options-for-esp8266

If you are using the old version esp8266 sdk, you should update.

closed 2023-03-04 by mobizt

JoergTiedemann, 2023-03-05:
Hi 
thanks for your answer, but I think there must be an issue. I‘ve updated to newest core version but same behavior. My application works for more then one year without any memory leaks or chrash reboot. But if I use the firebase library and i got an error „response payload read timed out“ while writing and bad WiFi connection, I see that the free heap is going down, so I think there is a memory leak in the lib in this case.
Everything works fine if I have free heap from 10,5KB to 11KB, but if heap is going below 8,5KB it chrashed in a very short time.
What is the minimum required free heap and is there any possibility to minimize this ? Could I use http instead of https connections to firebase or could I delete the actual instance of the library and generate a complete new one at runtime if I detect any errors ?
Thanks for your help in forward
Regards joerg

mobizt, 2023-03-05:
The library was well managed. It is not possible at the current release for memory leaks.

Your device free heap is very critically low which I will never run my esp8266 device in this extremely low memory like this.

Your device should have at least 15-20k when you are using SSL client application.  Only 10k is impossible to make SSL client works and the system tasks may fail internally.

You should think again in this issue.

JoergTiedemann, 2023-03-05:
Is there any way to use http instead of https connection so ssl is not required ?

mobizt, 2023-03-05:
I replied in new post.