mobizt/Firebase-ESP8266 #357: ESP8266 is crashing while writing to firebase realtime database

JoergTiedemann, 2023-03-05:
Hi 
Referred to #356 thanks for your answer, but I think there must be an issue. I‘ve updated to newest core version but same behavior. My application works for more then one year without any memory leaks or chrash reboot. But if I use the firebase library and i got an error „response payload read timed out“ while writing and bad WiFi connection, I see that the free heap is going down, so I think there is a memory leak in the lib in this case.
Everything works fine if I have free heap from 10,5KB to 11KB, but if heap is going below 8,5KB it chrashed in a very short time.
What is the minimum required free heap and is there any possibility to minimize this ? Could I use http instead of https connections to firebase or could I delete the actual instance of the library and generate a complete new one at runtime if I detect any errors ?
Thanks for your help in forward
Regards joerg

mobizt, 2023-03-05:
The library was intensively test 24/7 for years. You should reduce memory usage and should manage your code to use memory efficiently.

You can enable debug options in Arduino IDE for OOM, WiFi and SSL which you will understand what I trying to tell.

closed 2023-03-05 by mobizt

mobizt, 2023-03-05:
When your device has only 10k free memory, it can work in case of non-dynamic data from sensor or from general application but not enough for SSL client which consumed much more memory not only dynamic data from server response payload but included the SSL session data to handle.

JoergTiedemann, 2023-03-05:
What is the minimum required free heap and is there any possibility to minimize this ? 

Could I use http instead of https connections to firebase or 

could I delete the actual instance of the library and generate a complete new one at runtime if I detect any errors ?

mobizt, 2023-03-05:
I replied to you already for SSL client. You can run all your code at extremely low memory in non-WiFi and interrupt disabled application. 

All Google services work through secure socket layer protocol or no plain text or http supports.

You can define FirebaseData object in your usage scope e,g, in function which all resource will free after exit the scope unless exit by interrupt.

With local usage of `FirebaseData` object, the server connection will be closed/opened when you want to read/send data, but it takes additional time for SSL handshake during sending the request.

Additionally, Stream will not work as it required keep alive connection for SSE method which available only when `FirebaseData` object defined globally.

mobizt, 2023-03-05:
I recommend you move to Raspberry Pi Pico for more available memory.

The same code can run smoothly in Pico due to it uses `BearSSL` engine in sdk as in ESP8266.