mobizt/Firebase-ESP8266 #363: FirebaseDataObject.clear() is not working

0111101010110, 2023-04-21:
Currently, I am using Firebase ESP Client version 4.3.9. 

This is my code:
#include <Firebase_ESP_Client.h>

FirebaseData FirebaseDataObject;

void do_a_func()
{
	if(do something successfully)
	{
		FirebaseDataObject.setBSSLBufferSize(2048 /* Rx buffer size in bytes from 512 - 16384 */, 2048 /* Tx buffer size in bytes from 512 - 16384 */);
		FirebaseDataObject.setResponseSize(2048);
	}
	else
	{
		// Do other thing
	}
}

void setup()
{
	// do something here
	do_a_func();
}

void loop()
{

	FirebaseJson SensorDataContent;
	
	SensorDataContent.set("fields/timestamp/timestampValue", safe_str_timestamp.c_str());
	FirebaseDataObject.pauseFirebase(false);
	if (Firebase.ready()) 
	{
		if(&FirebaseDataObject, Firebase_project_name, "", "fixed/document/path", SensorDataContent.raw())) 
		{
		  Serial.println(" >> Save successfully.");
		}

		else 
		{
		  Serial.println(FirebaseDataObject.errorReason());
		}
	}
	FirebaseDataObject.pauseFirebase(true);
        FirebaseDataObject.clear();

	SensorDataContent.clear()
	SensorDataContent.set("fields/string/stringValue", "random string");
	FirebaseDataObject.pauseFirebase(false);
	if (Firebase.ready()) 
	{
		if(&FirebaseDataObject, Firebase_project_name, "", "random/document/path", SensorDataContent.raw())) 
		{
		  Serial.println(" >> Save successfully.");
		}

		else 
		{
		  Serial.println(FirebaseDataObject.errorReason());
		}
	}
	FirebaseDataObject.pauseFirebase(true);
        FirebaseDataObject.clear();
	
	
	
	FirebaseDataObject.pauseFirebase(false);
	if(Firebase.ready())
	{
		if (Firebase.Firestore.deleteDocument(&FirebaseDataObject, Firebase_project_name, "", "random/document/path"))
		{
		  Serial.println(" >> Delete successfully.");
		}
		else
		{
		  Serial.println(FirebaseDataObject.errorReason());
		}
	}
	FirebaseDataObject.pauseFirebase(true);
        FirebaseDataObject.clear();
}

The output I get is:
>> Save successfully.
>> Save successfully.
>> Delete successfully.

Then: 
Document already exists: projects/testFirebase/databases/(default)/documents/fixed/document/path
>> Save successfully.
>> Delete successfully.

After 3 ~ 8 hours:
Document already exists: projects/testFirebase/databases/(default)/documents/fixed/document/path
Document already exists: projects/testFirebase/databases/(default)/documents/fixed/document/path
Document already exists: projects/testFirebase/databases/(default)/documents/fixed/document/path

It cause my task 2 and 3 fail. Why?

mobizt, 2023-04-21:
You are misunderstanding in the clear function. It used for library internal usage memory releasing.

closed 2023-04-21 by mobizt

0111101010110, 2023-04-21:
I am very new to the NodeMCU, so is there any method to overcome this?

mobizt, 2023-04-21:
You have the problem with your programming logic.

This line is not complete or you are accidentally erased something before posting.

`if(&FirebaseDataObject`, 

But with the error returns from server indicates that you are trying to create the document that is already existed.

0111101010110, 2023-04-21:
Noted. I will fix and try again.