mobizt/Firebase-ESP8266 #375: compiling in arduino json

demoking23, 2023-08-30:
In file included from c:\Users\ERSHITHVARAN\Documents\Arduino\libraries\FirebaseArduino\src/FirebaseArduino.h:22,
                 from C:\Users\ERSHITHVARAN\AppData\Local\Temp\.arduinoIDE-unsaved2023730-6176-3ewnot.vfdek\FirebaseStream_ESP8266\FirebaseStream_ESP8266.ino:20:
c:\Users\ERSHITHVARAN\Documents\Arduino\libraries\FirebaseArduino\src/Firebase.h:26:10: fatal error: ArduinoJson.h: No such file or directory
   26 | #include <ArduinoJson.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1

demoking23, 2023-08-30:
please tell me the suitable version for json library

mobizt, 2023-08-30:
It is not my library.

You should post it here.
https://github.com/FirebaseExtended/firebase-arduino

closed 2023-08-30 by mobizt