mobizt/Firebase-ESP8266 #382: PLEASE HELP

novanmizan, 2023-11-04:
I have a problem with my code
i get errors 
error: 'class Firebase_ESP_Client' has no member named 'pushInt'
error: 'class Firebase_ESP_Client' has no member named 'failed'

// Include Library Arduino
#include <Arduino.h>
#include "PMS.h"
#include <MQUnifiedsensor.h>
#include <WiFi.h>
#include <Firebase_ESP_Client.h>
#include "addons/TokenHelper.h"
#include "addons/RTDBHelper.h"

// Connect To wifi And Setup Firebase
#define WIFI_SSID "test"
#define WIFI_PASSWORD "test"
#define PROJECT_ID "test"
#define API_KEY "test"

// Pin Alarm
#define buzzer 2
#define LED 4

//DHT
#include "DHT.h"
DHT dht(15, DHT22);
float t;
float h;

// Configuration Firebase
FirebaseData fbdo;
FirebaseAuth auth;
FirebaseConfig config;
bool signupOK = false;

//konfigurasi lcd
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include <Adafruit_Sensor.h>
#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels
#define OLED_RESET -1
#define ID 002
Adafruit_SSD1306 display(SCREEN_WIDTH,SCREEN_HEIGHT,&Wire,OLED_RESET);

// Konfigurasi PMS
PMS pms(Serial2);
PMS::DATA data;
float particulate;
int pm2;

// Konfigurasi Sulfur
#define placa "ESP32"
#define Voltage_Resolution 3.3
#define pin 34 //Analog input 0 of your arduino
#define type "MQ-136" //MQ136
#define ADC_Bit_Resolution 12 // For arduino UNO/MEGA/NANO
#define RatioMQ136CleanAir 3.6//RS / R0 = 3.6 ppm  
float sulfur;
MQUnifiedsensor MQ136(placa, Voltage_Resolution, ADC_Bit_Resolution, pin, type);

// Konfigurasi Ammonia
#define ammonia 35
float m = -0.263; // Slope
float b = 0.42; // Y-Intercept
float VRL; // Define variable for sensor voltage 
float RS_air; // Define variable for sensor resistance
float R0 ; // Define variable for R0
float RL = 47;
float ratio; // Define variable for ratio
float sensorValue; // Define variable for analog readings 
float ppm;
int baca;
float analog_value;
float VRL1;
float Rs1;

//konfigurasi rtos
TaskHandle_t mainTask; //nama bebas
TaskHandle_t DisplayTask; //nama bebas

// Konfigurasi Data 
String documentPath = "sensors/Graha/logData"; // Ganti dengan nama sensor yang sesuai
FirebaseJson content;

void sendToFirestore(float sulfur, float particulate, float ppm, float h, float t) {
  // Membuat objek FirebaseJson untuk menampung data
  FirebaseJson dataJson;
  dataJson.set("timestamp", "2023-10-19 14:30:00");
  dataJson.set("humidity", h);
  dataJson.set("ammonia", ppm);
  dataJson.set("particulate", particulate);
  dataJson.set("sulfur", sulfur);
  dataJson.set("temperature", t);
  dataJson.set("humidityStatus", "Normal");
  dataJson.set("particulateStatus", "Low");

   // Mengirim data ke Firestore
  String logDataId = Firebase.pushInt(fbdo, PROJECT_ID, documentPath.c_str(), dataJson);
  if (Firebase.failed(fbdo)) {
    Serial.print("Gagal mengirimkan data ke Firestore: ");
    Serial.println(fbdo.errorReason());
  } else {
    Serial.print("Data berhasil dikirim ke Firestore dengan ID: ");
    Serial.println(logDataId);
  }
}

// Calibrate RO one time ammonia sensor
void Sensor1() {
  for (int test_cycle = 1; test_cycle <= 500; test_cycle++) {
    analog_value = analog_value + analogRead(ammonia);
  }
  analog_value = analog_value / 500.0;
  VRL1 = analog_value * (3.3 / 4095.0);
  Rs1 = ((3.3 / VRL1) - 1) * RL;
  R0 = Rs1 / 3.6;
}

void programCore0(void *pvParameters) {
  for (;;) {
    // Pembacaan Partikulate
    pms.read(data);
    particulate = data.PM_AE_UG_2_5;
    pm2 = data.PM_AE_UG_2_5;
    // Pembacaan Sulfur
    MQ136.update();
    sulfur = MQ136.readSensor();
    // Pembacaan Ammonia
    VRL = analogRead(ammonia) * (3.3 / 4095.0);
    RS_air = ((3.3 * RL) / VRL) - RL;
    ratio = RS_air / R0;
    ppm = pow(10, ((log10(ratio) - b) / m));
   //DHT
  t = dht.readTemperature();
   h = dht.readHumidity();
 delay(100);
    // Kirim data ke Firestore
    sendToFirestore(sulfur, particulate, ppm, t, h);
  }
}

void programCore1(void *pvParameters){ //ini fungsi loop yang jalan di core 0
  for(;;){
  // Display LCD 
  display.clearDisplay();  //clear display
  display.setTextSize(1);   // display temperature
  display.setCursor(10,3);
  display.print("Temperature ");
  display.setTextSize(1);
  display.setCursor(85,3);
  display.print(t);

  display.setTextSize(1);   // display humidity
  display.setCursor(10, 17);
  display.print("Humidity ");
  display.setTextSize(1);
  display.setCursor(85, 15);
  display.print(h);

  display.setTextSize(1);     // display NH3
  display.setCursor(10, 27);
  display.print("NH3 ");
  display.setTextSize(1);
  display.setCursor(85, 27);
  display.print(ppm);

  
  display.setTextSize(1);     // display sulfur
  display.setCursor(10, 37);
  display.print("Sulfur ");
  display.setTextSize(1);
  display.setCursor(85, 37);
  display.print(sulfur);
 

 display.setTextSize(1);     // display sulfur
  display.setCursor(10, 47);
  display.print("PMS ");
  display.setTextSize(1);
  display.setCursor(85, 47);
  display.print(pm2);
  display.display(); 
  }
  
}

void setup() {
  // Konfigurasi Display
  Serial.begin(9600);
  Serial2.begin(9600);
  // Konek Firebase
  Serial.println("OK");
  // Koneksi ke WiFi
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting To Wifi ");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.println("Tidak terhubung ke WiFi...");
    delay(300);
  }
  Serial.print("Terhubung ke WiFi...");
  Serial.println(WiFi.localIP());

  //
  config.api_key = API_KEY;

  if (Firebase.signUp(&config, &auth, "", "")) {
    Serial.print("OK");
    signupOK = true;
  } else {
    // Serial.printf("%s\n", config.signer.signupError.message.c_str());
  }
  config.token_status_callback = tokenStatusCallback;
  // Inisialisasi Firebase
  Firebase.begin(&config, &auth);
  Firebase.reconnectWiFi(true);
  Serial.println("Terhubung ke Firebase...");
   digitalWrite (LED,HIGH);
  // LCD Set
  display.begin(SSD1306_SWITCHCAPVCC, 0x3C);
  delay(2000);
  display.clearDisplay();
  display.setTextColor(WHITE);
 display.display();

  // Konfigurasi PinMode
  pinMode(ammonia, INPUT);
  pinMode(pin, INPUT);
  pinMode(buzzer,OUTPUT);
  pinMode(LED,OUTPUT);

  // Konfigurasi Sulfur dan Ammonia
  Sensor1();
  MQ136.setRegressionMethod(1);
  MQ136.setA(36.737);
  MQ136.setB(-3.536);
  MQ136.init();
  float calcR0 = 3.6;
  for (int i = 1; i <= 10; i++) {
    MQ136.update();
    calcR0 += MQ136.calibrate(RatioMQ136CleanAir);
    Serial.print(".");
  }
  MQ136.setR0(calcR0 / 10);

  // Konfigurasi Core to Core
  xTaskCreatePinnedToCore(
    programCore0,
    "mainTask",
    10000,
    NULL,
    1,
    &mainTask,
    0
  );
  xTaskCreatePinnedToCore(
    programCore1, //nama bebas, ini nama fungsinya
    "DisplayTask", //sesuaikan sama nama yang di "TaskHandle_t"
    10000,
    NULL,
    1,
    &DisplayTask,
    1 //ini core yang mau dipake
  );
}

void loop() {
  vTaskDelete(NULL);
}

mobizt, 2023-11-05:
https://github.com/mobizt/Firebase-ESP8266/tree/master/examples

closed 2023-11-05 by mobizt

novanmizan, 2023-11-05:
Why my pushint error sir ? @mobizt

mobizt, 2023-11-05:
https://github.com/mobizt/Firebase-ESP8266#append-data