mobizt/Firebase-ESP8266 #388: HELP : Executed HTTPS and connection Firebase

Ricardo-Manzanares, 2024-03-03:
Hi devs,

On my ESP8266 Firebase connections work correctly, also HTTPS requests to external custom API.

The problem I have is when the Firebase connection is initialized in the setup or post startup functions, any HTTPS request that is executed returns a -1.

If I disable the Firebase connection, just by commenting out the Firebase.begin(&config, &auth) line; the HTTPS requests work correctly.

Is there any incompatibility of using both connections?

My ESP8266 receives an HTTP connection on an ESP8266WebServer, queries data to Firebase and executes an HTTPS connection to a custom API and I have the problem that the last part of the process returns -1.

any idea?

thanks

Translated with DeepL.com (free version)

mobizt, 2024-03-03:
You should not open issue in such case.

There is nothing related between your code and library.

mobizt, 2024-03-03:
If you debug the WiFi, OOM and SSL in Arduino debug options, you will see why your code doesn't work.

Please search Google for how to debug your ESP8266 in Arduino IDE.

FYI.
The discussions section is for posting question and help request.

closed 2024-03-03 by mobizt