mobizt/Firebase-ESP8266 #390: HELP :: captivePortal and firebase connection

Clonethis, 2024-03-05:
I want to create a device that can connect to any wifi using the default esp8266 captive portal. While I am trying to compile those two libraries, it gets broken. I didn't find anything similar to the mentioned error messages in the code. 

```cpp error
/User/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /private/var/folders/4q/czz88n8x79v3kgkz0h4gxvfm0000gn/T/arduino/sketches/0BD8CAFF987449BBD7AD03355E2565F3/sketch/objs.a(esp8266_firebase.ino.cpp.o): in function `_ZN16esp8266webserver24ESP8266WebServerTemplateI10WiFiServerE12handleClientEv':
/User/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h:386: undefined reference to `_Z11connectWifiv'
/User/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /User/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h:400: undefined reference to `_Z11connectWifiv'
collect2: error: ld returned 1 exit status
Multiple libraries were found for "SD.h"
  Used: /User/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/SD
  Not used: /User/Library/Arduino15/libraries/SD

```

Is it possible to set up a captive portal and also connect it with Firebase? Thank you in advance.

closed 2024-03-05 by mobizt

Clonethis, 2024-03-05:
Thank you, works like a charm. (Also I am really sorry for that beginner mistake) How did you come up with that solution, the message is quite unclear and points somewhere else. (The main mistake was different case - so typo). 

Why would compiler not throw error on typo and rather show some arbitrary message?

mobizt, 2024-03-05:
> /User/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: [...]

You can see the recognized word `connectWifi` from `_Z11connectWifiv` from the compiler error above.

You can try by declaring any function prototype without the real function body, calling that function from anywhere and see the compiler result.

Clonethis, 2024-03-06:
Thank you for explanation. Makes sense now. 

Wish you the best of luck in programming :)!