mobizt/Firebase-ESP8266 #391: BUG FirebaseESP8266.h with version @^4.4.12 of Library

Marek128b, 2024-03-15:
**Describe the bug**
the function dose not work with esp8266 as the FirebaseESP8266.h file does not exist. error in addon RTDBHelper.h

`#elif defined(ESP8266) || defined(MB_ARDUINO_PICO)
#include <FirebaseESP8266.h>
#endif`

**To Reproduce**
Steps to reproduce the behavior:
1. have a ESP8266 `d1_mini_lite`
1. include `#include <addons/RTDBHelper.h>`
2. See error

**Expected behavior**
no problems to import RTDBHelper.h

**Screenshots**
in Main.cpp
[image]

in addons/RTDBHelper.h
[image]

in PlatfomIO.ini
[image]

**IDE and its version:**
 - PlatformIO
 - Version [6.1.13]

**Additional context**
nothing more

mobizt, 2024-03-15:
Error is on your side instead and library works normally.

mobizt, 2024-03-15:
Check your library installation.
https://github.com/mobizt/Firebase-ESP8266/blob/master/src/FirebaseESP8266.h

Don't install library on network drive e.g. OneDrive.

mobizt, 2024-03-15:
For PIO, you have to clean the project when you edit the platformio.ini.

closed 2024-03-15 by mobizt

mobizt, 2024-03-15:
FYI.

All my libraries developed in PlatformIO and test for both PlatformIO and Arduino.