mobizt/Firebase-ESP8266 #392: FirebaseJson is confilcted with json inside ESP8266WebServer library

b1n0-kun, 2024-03-25:
Hi Mobitz.

Today I bumped into a issue of "json" being declared in ESP8266WebServer library when trying to use both Firebase and ESP8266 Webserver inside one project.

Can you give me a hint on how to work around or it's impossible to use both library together ?

Here's is part of the Output log during Verification:

c:\Users\Linh\Documents\Arduino\libraries\Firebase_ESP8266_Client\src/addons/RTDBHelper.h:43:9: error: 'FirebaseJson' was not declared in this scope; did you mean 'Firebase'?
   43 |         FirebaseJson *json = data.to<FirebaseJson *>();
      |         ^~~~~~~~~~~~
      |         Firebase
c:\Users\Linh\Documents\Arduino\libraries\Firebase_ESP8266_Client\src/addons/RTDBHelper.h:43:23: error: 'json' was not declared in this scope; did you mean 'mime::json'?
   43 |         FirebaseJson *json = data.to<FirebaseJson *>();
      |                       ^~~~
      |                       mime::json
In file included from C:\Users\Linh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:32,
                 from C:\Users\Linh\Documents\Arduino\FS_Browser_WebSocket_Email\FS_Browser_WebSocket_Email.ino:43:
C:\Users\Linh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/detail/mimetable.h:17:3: note: 'mime::json' declared here
   17 |   json,
      |   ^~~~
In file included from C:\Users\Linh\Documents\Arduino\FS_Browser_WebSocket_Email\Firebase.ino:5:
c:\Users\Linh\Documents\Arduino\libraries\Firebase_ESP8266_Client\src/addons/RTDBHelper.h:43:35: error: parse error in template argument list

Thank you in advance

b1n0-kun, 2024-03-25:
Hi Mobitz,

Thank you for the quick responding,
After asking you the question, I gged a little bit more and tried to include Firebase library before Webserver and somehow the issue disappeared.. But your test with Firebase included later still worked..hmm..It's hard to understand about the root cause.
But thanks anyway !!!

mobizt, 2024-03-25:
I'm not sure this related to your issue or not.
I found some problem with FirebaseJson integration when anywhere in the code or other libraries used the statement `"using namespace std"`
I update the FirebaseJson to fix the issue I've found, you can try by copy this [json](https://github.com/mobizt/Firebase-ESP8266/tree/master/src/json) folder in Github and replace with `json` folder in your your PC.

b1n0-kun, 2024-03-25:
Thanks Mobitz, will try to replace and test again when I have time !!!
I'm squeezing out the memory of ESP with Webserver,FSbrowser, WebSocket, Email and Firebase :D 
So far the device still function well !

Hope to get your support from you in the future also.
Best regards

jiperez11, 2024-03-25:
Thanks Team

closed 2024-03-25 by mobizt