mobizt/Firebase-ESP8266 #396: Wi-Fi Reset Doesn't Initialize Firebase Variables and LED Status Doesn't Update Post-Reconnection

abdulsamadilyas82, 2024-04-27:
After resetting the Wi-Fi connection and reconnecting, the device fails to initialize the `ledDb` variable in Firebase with the value of 0. Additionally, it does not update the LED status based on changes made in Firebase after reconnecting to Wi-Fi.

Expected Behavior:

1. Upon reconnecting to Wi-Fi, the device should initialize the `ledDb` variable in Firebase with a value of 0.
2. The device should continuously monitor changes in the `ledDb` variable in Firebase and update the LED status accordingly, even after reconnecting to Wi-Fi.

Steps to Reproduce:

1. Connect the device to Wi-Fi and ensure it is properly interfacing with Firebase.
2. Reset the Wi-Fi connection.
3. Reconnect the device to Wi-Fi.
4. Observe that the `ledDb` variable in Firebase is not initialized with a value of 0, and LED status does not update based on changes in Firebase.

#include <Arduino.h>
#include <WiFi.h>
#include <BluetoothSerial.h>
#ifdef ESP32
#include <FirebaseESP32.h>
#endif

#ifdef ESP8266
#include <FirebaseESP8266.h>
#endif

// Define the Firebase credentials
#define FIREBASE_AUTH "AIzaSyBl1s-R3qn7VBRIQdJd_EgWXhOnu2kjYUY"
#define FIREBASE_HOST "https://nodemcu-bda0c-default-rtdb.firebaseio.com/"
#define EMAIL "abdulsamadilyas82@gmail.com"
#define PASSWORD "12345g_G"

// Create a Bluetooth Serial object
BluetoothSerial SerialBT;
FirebaseData firebaseData;
FirebaseConfig config;
FirebaseAuth auth;

const int led = 13;

void setup() {
  Serial.begin(115200);
  pinMode(led, OUTPUT);

  // Start Bluetooth Serial
  SerialBT.begin("ESP32_BT"); // Bluetooth device name
  delay(500);
  Serial.println("The device started, now you can pair it with Bluetooth!");

  // Initialize Wi-Fi and Firebase
  initializeWiFiAndFirebase();
}

void loop() {
  if (SerialBT.available()) {
    String command = SerialBT.readString();
    command.trim();
    if (command == "reset") {
      // Disconnect from current Wi-Fi
      WiFi.disconnect(true);
      Serial.println("Disconnected from Wi-Fi");

      // Ask for new Wi-Fi SSID and password
      initializeWiFiAndFirebase();

    }
  }

  // Delay before retrieving LED status from Firebase
  delay(500);

  // Retrieve LED status from Firebase
  Firebase.getInt(firebaseData, "/ledDb");
  int fireStatus = firebaseData.to<int>();  
  Serial.println(fireStatus);
  
  if (fireStatus == 0) {
    Serial.println("LED Turned OFF");
    digitalWrite(led, LOW);
  } else if (fireStatus == 1) {
    Serial.println("LED Turned ON");
    digitalWrite(led, HIGH);
  } else {
    Serial.println("Unknown LED status received");
  }

  delay(1500); // Delay after each loop iteration
}

String getData() {
  String ssidAndPassword = "";
  SerialBT.println("Send SSID and Password in the format SSID:PASSWORD");
  while (!SerialBT.available()) {
    delay(20); // Wait for input
  }
  Serial.println("Getting  SSID & PASSWORD");
  while (SerialBT.available()) {
    ssidAndPassword += char(SerialBT.read());
    delay(20); // Wait for all data to come in
  }

  SerialBT.println(ssidAndPassword + "Sent");
  return ssidAndPassword;
}

void initializeWiFiAndFirebase() {
  String ssidAndPassword = getData();
  ssidAndPassword.trim();

  int colonIndex = ssidAndPassword.indexOf(':');
  if (colonIndex != -1) {
    String ssid = ssidAndPassword.substring(0, colonIndex);
    String password = ssidAndPassword.substring(colonIndex + 1);

    ssid.trim();
    password.trim();

    Serial.print("Attempting to connect to Wi-Fi with SSID: ");
    Serial.println(ssid);

    WiFi.begin(ssid.c_str(), password.c_str());
    int attempts = 0;
    while (WiFi.status() != WL_CONNECTED && attempts < 20) {
      Serial.println("Attempting to connect to Wi-Fi...");
      delay(500);
      attempts++;
    }

    if (WiFi.status() == WL_CONNECTED) {
      Serial.println("Connected to Wi-Fi");
      Serial.print("IP Address: ");
      Serial.println(WiFi.localIP());

      config.api_key = FIREBASE_AUTH;
      auth.user.email = EMAIL;
      auth.user.password = PASSWORD;
      config.database_url = FIREBASE_HOST;

      Firebase.begin(&config, &auth);

      // Initialize ledDb variable in Firebase every time the device connects to Wi-Fi
      Firebase.setInt(firebaseData, "/ledDb", 0);

      Serial.println("Firebase initialized");
    } else {
      Serial.println("Failed to connect to Wi-Fi");
      SerialBT.println("Failed to connect to Wi-Fi");
      delay(5000); // Wait before retrying
      initializeWiFiAndFirebase(); // Retry connection
    }
  } else {
    Serial.println("Invalid SSID and password format. Please send SSID:PASSWORD.");
    SerialBT.println("Invalid SSID and password format. Please send SSID:PASSWORD.");
    delay(5000); // Wait before retrying
    initializeWiFiAndFirebase(); // Retry connection
  }
}

mobizt, 2024-04-28:
You should follow the library examples.

Your code shows that you are using library incorrectly.

closed 2024-04-28 by mobizt

abdulsamadilyas82, 2024-04-28:
> You should follow the library examples.
> 
> Your code shows that you are using library incorrectly.

Can you show me which example.

mobizt, 2024-04-28:
Are you new to Arduino and this library?

This library was developed and maintain many years.

mobizt, 2024-04-28:
And you should read the Readme that this library will be deprecated this year and the new library is recommended.

You should check all examples of this library if you are continuing to use it.

The examples in this repo was removed recently before the deprecation announcement.

The examples are still here for the latest release https://github.com/mobizt/Firebase-ESP8266/tree/v4.4.14/examples and in your Arduino libraries folder.

abdulsamadilyas82, 2024-05-09:
Hy, I just want to ask can we do the firebase connection without auth like
without email password.....i can simply do this  Firebase.begin(&config);

mobizt, 2024-05-09:
Check the library examples which are included in the library installation folder.