mobizt/Firebase-ESP8266 #397: ERROR CODE: -3, REASON: NOT CONNECTED  STREAM TIMED OUT, RESUMING...

NGUYENVANY04, 2024-05-18:
Hello, I use your library for work a lot. But recently I need to listen for change events from realtime database. and I encountered this error, the problem is that sometimes it works fine, other times it logs the above error. I look forward to your help. and below is my code : 
void connect_wifi()
{
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(1000);
    Serial.print(".");
  }
  Serial.println("\nConnected to WiFi");
  Serial.print("WiFi Signal Strength: ");
  Serial.println(WiFi.RSSI());
}

void connect_firebase()
{
  Firebase.begin(LINK, SECRETS);
  Firebase.reconnectWiFi(true);

  if (Firebase.ready())
  {
    Serial.println("Connected to Firebase");
  }
  else
  {
    Serial.println(fbdo.errorReason());
  }
}
// Global function that handles stream data
void streamCallback(StreamData data)
{
  Serial.println("Data changed, turning on the light.");
  if (!state_light)
  {
    digitalWrite(DEVICE, LOW); // Turn on the light
    state_light = true;        // Update the light state to ON
  }
  lastDataChangeMillis = millis(); // Update the last data change time
}

// Global function that notifies when stream connection is lost
// The library will resume the stream connection automatically
void streamTimeoutCallback(bool timeout)
{
  if (timeout)
    Serial.println("stream timed out, resuming...\n");

  if (!fbdo.httpConnected())
    Serial.printf("error code: %d, reason: %s\n\n", fbdo.httpCode(), fbdo.errorReason().c_str());
}

void handle_stream_data_firebase()
{
  while (!Firebase.beginStream(fbdo, "/system/table_1/realtime"))
  {
    // Could not begin stream connection, then print out the error detail
    Serial.println(fbdo.errorReason());
    delay(1000); // Wait for a while before retrying
  }
  Firebase.setStreamCallback(fbdo, streamCallback, streamTimeoutCallback);
}

void setup()
{
  Serial.begin(115200);
  pinMode(DEVICE, OUTPUT);
  digitalWrite(DEVICE, HIGH); // Initialize the device to be off

  connect_wifi();
  connect_firebase();
  handle_stream_data_firebase();
}

void loop()
{
  if (millis() - lastDataChangeMillis > TIME_RESTART_LOOP)
  {
    lastLoopMillis = millis();
    if (!Firebase.ready())
    {
      lastDataChangeMillis = millis();
      Serial.print(fbdo.errorReason());
    }
    // Check if the light should be turned off due to inactivity
    if (Firebase.ready())
    {
      if (state_light && (millis() - lastDataChangeMillis > TIMEOUT_DURATION))
      {
        Serial.println("No data change detected, turning off the light.");
        digitalWrite(DEVICE, HIGH); // Turn off the light
        state_light = false;        // Update the light state to OFF
      }
    }
  }
}
thank you very much

mobizt, 2024-05-18:
This is incorrectly usage.

```cpp
while (!Firebase.beginStream(fbdo, "/system/table_1/realtime"))
{
// Could not begin stream connection, then print out the error detail
Serial.println(fbdo.errorReason());
delay(1000); // Wait for a while before retrying
}
```

See the examples for how to use library properly.

closed 2024-05-18 by mobizt

NGUYENVANY04, 2024-05-18:
void handle_stream_data_firebase()
{
  if (!Firebase.beginStream(fbdo, "/system/table_1/realtime"))
  {
    // Could not begin stream connection, then print out the error detail
    Serial.println(fbdo.errorReason());
  }
  Firebase.setStreamCallback(fbdo, streamCallback, streamTimeoutCallback);
} 
Thank you, I have edited the code to look like this. But why when esp32 just started, after "Connected to Firebase" it says "Data changed, turning on the light." while real-time data does not change

NGUYENVANY04, 2024-05-18:
During testing, I encountered many errors. I hope you can help. Thanks a lot
[ 10209][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for manager-bida-default-rtdb.asia-southeast1.firebasedatabase.app
connection refused
[ 18242][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for manager-bida-default-rtdb.asia-southeast1.firebasedatabase.app
stream timed out, resuming...

error code: -3, reason: not connected

mobizt, 2024-05-18:
It is because of the internet problem.

NGUYENVANY04, 2024-05-18:
> void handle_stream_data_firebase() { if (!Firebase.beginStream(fbdo, "/system/table_1/realtime")) { // Could not begin stream connection, then print out the error detail [...]

As for this problem, can you help me?

mobizt, 2024-05-18:
It is library behavior for years. Did you never notice?

The stream response will be available since it was established because it is http get request in stream mode.

This library is obsoleted and will not support for help after this year.

You can move to use [new library](https://github.com/mobizt/FirebaseClient) but it required the whole code changes

NGUYENVANY04, 2024-05-18:
Is the library's new stream processor good? . Thank you

mobizt, 2024-05-18:
Read the library documentation thoroughly before use.